Nerivec/silabs-firmware-builder #3: LQI signal strength 0 after updating Sonoff ZBDongle-E to 8.0.1

quensen, 2024-08-19:
I updated my _ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2_ to firmware `sonoff_zbdonglee_ncp-uart-sw_115200_8.0.1.0.gbl` from `https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.1`.
Since then, all newly received signal strengths show 0 LQI.
[image]
[image]

After switching back to 7.4.3.0 (`ncp-uart-hw-v7.4.3.0-zbdonglee-230400.gbl`) everything is back to normal.
[image]

Nerivec, 2024-08-22:
As mentioned [here](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussion-6887141), LQI reporting in messages is currently broken in-firmware. It should be fine in the network map however. Silabs was notified.

Nerivec, 2024-09-22:
Should be fixed in 8.0.2.

closed 2024-09-22 by Nerivec