Nerivec/silabs-firmware-builder #5: New release v2024.6.2 : Which file I need for my SLZB-06M with ember driver?

MilesTEG1, 2024-09-22:
Hello,
I installed severals weeks ago this file: smlight_slzb06m_ncp-uart-sw_115200_8.0.1.0.gbl from a branch you had since. (I don't find this file anymore here)

Your new release v2024.6.2 has twi files for my SLZB-06M :
- [smlight_slzb06m_ncp-uart-sw_115200_8.0.2.0.gbl](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2/smlight_slzb06m_ncp-uart-sw_115200_8.0.2.0.gbl)
- [smlight_slzb06m_rcp-uart-802154-blehci-sw_460800_2024.6.2.gbl](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2/smlight_slzb06m_rcp-uart-802154-blehci-sw_460800_2024.6.2.gbl)

Which one I have to flash? Is it compatible with ember driver on Z2M?

Thanks in advance.

quensen, 2024-09-22:
@MilesTEG1 
NCP is Zigbee
RCP is Thread

MilesTEG1, 2024-09-23:
> @MilesTEG1 
> 
> NCP is Zigbee
> 
> RCP is Thread

Ok! Thank you 😊

The question of the Ember driver remains. Is this firmware compatible with ember?

quensen, 2024-09-23:
> The question of the Ember driver remains. Is this firmware compatible with ember?

I have a different coordinator dongle, but ialso Zigbee2MQTT and installed the relevant 8.0.2 version successfully yesterday and it’s working fine with Z2M so far.

MilesTEG1, 2024-09-23:
> > The question of the Ember driver remains. Is this firmware compatible with ember?
> 
> I have a different coordinator dongle, but ialso Zigbee2MQTT and installed the relevant 8.0.2 version successfully yesterday and it’s working fine with Z2M so far.

Hello,
So, you use Ember as Zigbee driver, right?

quensen, 2024-09-23:
Yes, I have `adapter: ember` in my _configuration.yaml_

MilesTEG1, 2024-09-23:
> Yes, I have `adapter: ember` in my _configuration.yaml_
> 
> Z2M reports: `Coordinator-Typ EmberZNet` on the About page.

Yeah, that's what I wanted to knwo. Thank you very much.

MilesTEG1, 2024-09-23:
Flash done with smlight_slzb06m_ncp-uart-sw_115200_8.0.2.0.gbl.
No errors, and continue to works fine.
Thanks @Nerivec

quensen, 2024-09-23:
:-(

MilesTEG1, 2024-09-23:
@quensen You seem to be offended, but I already thank you for your answers.
In my last message, I was thanking @Nerivec for his version of the firmware for my SLZB-06M.
@quensen you were very helpful in answering my questions, thanks for that.

Nerivec, 2024-09-23:
Some more details: `802514-blehci` version is indeed RCP with support for ZigBee + OpenThread + BLE. This is highly experimental and requires a more advanced setup (there's an official [Silabs Docker container](https://hub.docker.com/r/siliconlabsinc/multiprotocol), and a few [repo](https://github.com/m33ts4k0z/z2m-ember-multiprotocol), [repo](https://github.com/Nerivec/silabs-multiprotocol-builder) that experiment with it).

As for the version, usually, considering a version `x.y.z`, as long as Z2M supports `x.y`, when `z` changes, it should automatically be compatible, as these are only fixes/improvements, not breaking changes.

closed 2024-09-23 by Nerivec