Nerivec/silabs-firmware-builder #7: Can’t Flash sonoff dongle e

wastez, 2024-10-13:
Hello,

I can’t flash the ncp 8.0.2 firmware on my sonoff dongle e. 
I always get the following error:
ValueError: 'ncp-uart-sw' is not a valid FirmwareImageType

I‘m using universal silabs flasher, also with force or allow cross flashing option, without success. 

I had a look in my old darxkst firmware and in yours. 
The old firmware has ncp-uart-hw even if it is a sw, but darxkst leave it as it is from silabs. 
So it is clear that it will not let it flash without the cross flashing option. 
But as I said I isn’t working. 

Are you sure everything is alright with the sonoff dongle e firmware?

lalek54, 2024-10-15:
Exactly the same problem here. 
migrating from darkxst firmware and same error: ValueError: 'ncp-uart-sw' is not a valid FirmwareImageType

Even trying to force flash is not possible.

mamrai1, 2024-10-15:
Same problem with universal silabs flasher. Managed to flash with ember-zli. Seems very stable

wastez, 2024-10-15:
@mamrai1 
Thank you for the tip, this new tool is really cool. 

Firmware is now running.

lalek54, 2024-10-16:
> Same problem with universal silabs flasher. Managed to flash with ember-zli. Seems very stable

Thanks a lot. Ember-zli is perfect.

closed 2024-10-17 by wastez

wastez, 2024-10-17:
The firmware is fast and rock solid. 
Really great. 
Long time test will show if the state remains but if it’s better then the v7 firmwares.