Nerivec/silabs-firmware-builder #9: Add Zigbee FW Version in the SLZB-06M dashboard

MilesTEG1, 2024-10-29:
Hello, @Nerivec 
I flashed the new Zigbee FW you plublished this morning.
And as always, when falshing custom FW for the SLZB-06M, there is no indication of the FW Version we flashed, or had before flash.

[image]

Is it possible to add some indication about the version you provide?
Like `Nerivec-v2024.6.2-update1` or `Nerivec-ncp_8.0.2.0_115200_sw_flow` or anything else that gives the information of what fw version we have?

Thanks in advance.

Nerivec, 2024-10-29:
From what I understand, this stuff is linked to the SMLight API call, not the content of the file being flashed. So, at least for now, this is not possible from my end.

https://github.com/smlight-tech/pysmlight/blob/8b9e220412f8fffc9d9dec32194fa4df47e4d039/pysmlight/web.py#L256

MilesTEG1, 2024-11-07:
@Nerivec 
Ok.
Do you know what it'll need to get your FW version to show?
I'm writing to the support, maybe they can add something in their API.

Tarik2142, 2024-11-07:
> From what I understand, this stuff is linked to the SMLight API call, not the content of the file being flashed. So, at least for now, this is not possible from my end.
> 
> https://github.com/smlight-tech/pysmlight/blob/8b9e220412f8fffc9d9dec32194fa4df47e4d039/pysmlight/web.py#L256

Yes, that's true.
Since CC and EFR use different firmware version formatting, we had to come up with a way to standardize this, so the standardized version is passed in the API request.
Of course, we do not know which version the user will flash in the case of custom firmware.

Nerivec, 2024-11-07:
@Tarik2142 I was talking about this with darkxst. For EFR, when processing the file, you could detect the GBL HEADER tag, and if present, then see if metadata is available, then use this, since versioning has been present for a while now in all builders.
https://github.com/Nerivec/ember-zli/blob/0aaf009159093cd7967ae229c7a0769ba747d127/src/utils/bootloader.ts#L271-L297
_Though, I believe your API call requires a few extra metadata, that might not all be available in the GBL... but maybe can be derived or mapped?_

Tarik2142, 2024-11-07:
> @Tarik2142 I was talking about this with darkxst. For EFR, when processing the file, you could detect the GBL HEADER tag, and if present, then see if metadata is available, then use this, since [...]

Thanks, I think I can add that. probably will use sdk_version converted to a number like 20240602 to match the zstack version formatting. baudrate and fw_type would also be helpful.

MilesTEG1, 2024-11-07:
> Thanks, I think I can add that. probably will use sdk_version converted to a number like 20240602 to match the zstack version formatting. baudrate and fw_type would also be helpful.

Yeah! Wonderful news 😁

MilesTEG1, 2024-12-02:
> @MilesTEG1 All new dev builds now support metadata https://smlight.tech/flasher/firmware/bin/slzb06x/core/slzb-06-v2.6.8.dev14-ota.bin

Yeah!
I'll try it.
Some days ago, I flashed `v2.6.8.dev9` :)

MilesTEG1, 2024-12-02:
@Tarik2142 Thank you very much:
[image]

When did you publish a new zigbee FW?
[image]

On which build of silabs is it? (if it's base on silabs...)

Tarik2142, 2024-12-02:
this is firmware on SDK 7.4.5

Nerivec, 2024-12-02:
@Tarik2142 Quick turn-around! Thanks for the feedback. 😉

closed 2024-12-02 by Nerivec

MilesTEG1, 2024-12-02:
> this is firmware on SDK 7.4.5

Ok thanks for the answer 😊

@Nerivec is your version of FW is based on the same sdk too?

Nerivec, 2024-12-02:
No, it's 8.0.2 using the newer simplicity_sdk  (same for new release from darkxst). 7.4.5 uses the previous gecko_sdk.
Silabs has moved to the new one but will keep maintaining fixes & such for the previous one for a while (per their docs).

MilesTEG1, 2024-12-03:
@Nerivec Thank you very much for those explanations :)