Nerivec/silabs-firmware-builder #18: problem with OTA updates

antst, 2024-11-21:
When I am trying to do OTA firmware updates, I always end up with this error
```
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-11-21 17:20:23] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Total frames: RX=1699, TX=2135
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   DATA frames : RX=1694, TX=433
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   DATA bytes  : RX=52446, TX=24903
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Retry frames: RX=3, TX=3
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1698
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Retry dupes     : RX=3
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2024-11-21 17:20:32] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=175 Len=10].
[2024-11-21 17:20:54] info: 	zh:ember:uart:ash: Port closed.
[2024-11-21 17:20:54] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-11-21 17:20:54] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-11-21 17:20:54] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-11-21 17:20:54] error: 	z2m: Adapter disconnected, stopping
```

I am using Z2M, but it does not look like Z2M problem (correct me if I am wrong). This log is from 8.02 update 2.
Sometimes I can get through 3% of update, sometimes through 7%...but end is always the same.

I have relatively dense network with 110 devices.

Nerivec, 2024-11-21:
Timeouts during OTA are not uncommon, it's a very taxing process... and very dependent on the quality of the network (the more depth, the more dependent).

When you say "dense", do you mean with lots of depth (as in "device > device > device > coordinator", that sort of map)?
How stable would you say your network is outside of OTA (any regularly occurring errors, etc.)?
Do you have any known "spammy" devices (like mmwave sensors)?

antst, 2024-11-22:
Lets start with fact that I switched to ember just few days ago, it was running with stack for years. And never had similar issues with OTA. It used to work.
Now, with ember. From one side I have a feeling (too little time to say for sure) that network is a bit more stable, in a sense that with zstack sometime I'd have temporary delays in communication with devices (particular kinds, mostly, I have real zoo). With ember it feels somehow better, but time will show. But over few days I haven't seen any errors in logs (aside of OTA), which is kind of  unusual after zstack.
But attempts to do OTA updates, in 100% of cases, fail like above.
I don't think I have particularly spammy devices, to be honest.
Topology wise, see attached image.
[image]

I still didn't re-pair few devices from original setup, so, technically, current network is smaller than it used to be and used to work.

antst, 2024-11-22:
Under different circumstances, I'd dive into code myself. I was quite familiar with zstack, and author of first cc2538 firmware, so no stranger to zigbee (and intrinsic illnesses of zstack). But at the moment I am trying to avoid diving into silabs stack, as it will eat quite some time :)

Nerivec, 2024-11-22:
When it throws that timeout error, it means it exhausted all possible retry steps, and then sync with host is lost. But silabs is indeed very aggressive with resetting the NCP... some places I already skipped it... But when the adapter sends an ERROR frame, we have no choice, it has to be handled as fatal, since it puts the NCP in failed state and it won't communicate again until a reset (RST).
I do think there is a weird behavior with `ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` (at least in some cases, seems to be related to depth), but I haven't been able to pinpoint it, we don't have access to the full source code for the firmware and I also can't replicate it myself... 😢

Can you pull out routers connected to green lines so we can better see the map, and uncheck "Siblings"?
Did you try updating end devices, routers? Close to coordinator, farther from coordinator? Behavior any different?
Any chance you could sniff an OTA up until it fails?

antst, 2024-11-22:
I tried to update routers, as I have no pending OTA for end devices. From what I see, proximity to coordinator does not matter. It all the same.  Map is coming

antst, 2024-11-22:
[image]

antst, 2024-11-22:
Interesting. Most of stuff is not connect with parent/child to coordinator.... But update to connected one (tried right now) failed also.

Nerivec, 2024-12-03:
I'm combining topics on this in this issue https://github.com/Koenkk/zigbee2mqtt/issues/23861 to keep track of it more easily.

closed 2024-12-03 by Nerivec