Nerivec/silabs-firmware-builder #19: router error

antst, 2024-12-03:
I run lates router FW, with converter per instructions.

And I regularly see this in logs
```
z2m: Failed to configure 'upstairs_router', attempt 1 (Error: Bind 0x84b4dbfffefceff7/1 genBasic from '0x8c65a3fffe54ba85/1' failed (Status 'TABLE_FULL')
```

I am not sure, is it supposed to be like that at the moment, or it is genuine error. But, just in case, here it is.

antst, 2024-12-04:
BTW, it would be lovely to have support for OTA updates for the router firmware.

Nerivec, 2024-12-04:
I assume the two IEEE addresses refer to router and coordinator?
Table full indicates the binding table of the device is full, bit strange in this scenario.
Could be some things didn't clear out properly from a previous network? Did you try clearing the NVM3 of the router (will require re-pairing)?

_OTA updates would require bootloader changes. It's easy enough to update that this is currently not worth the foreseeable troubles._

antst, 2024-12-04:
I just right bow tried to reflash and re-pair one which exists the issue. And this error comes out after paring immediately.
```
[2024-12-04 19:19:29] info: 	zh:controller: Succesfully interviewed '0x84b4dbfffefceff7'
[2024-12-04 19:19:29] info: 	z2m: Successfully interviewed '0x84b4dbfffefceff7', device has successfully been paired
[2024-12-04 19:19:29] info: 	z2m: Device '0x84b4dbfffefceff7' is supported, identified as: Silabs Silabs series 2 adapter with router firmware (Silabs series 2 router)
[2024-12-04 19:19:29] info: 	z2m: Configuring '0x84b4dbfffefceff7'
[2024-12-04 19:19:29] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x84b4dbfffefceff7/reset/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x84b4dbfffefceff7/set/reset","device":{"identifiers":["zigbee2mqtt_0x84b4dbfffefceff7"],"manufacturer":"Silabs","model":"Silabs series 2 adapter with router firmware (Silabs series 2 router)","name":"0x84b4dbfffefceff7","sw_version":"2024.6.2_Nerivec","via_device":"zigbee2mqtt_bridge_0x8c65a3fffe54ba85"},"enabled_by_default":false,"name":"Reset","object_id":"0x84b4dbfffefceff7_reset","options":["reset"],"origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"unique_id":"0x84b4dbfffefceff7_reset_zigbee2mqtt"}'
[2024-12-04 19:19:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Silabs series 2 adapter with router firmware","exposes":[{"access":2,"description":"Resets and launches the bootloader for flashing. If USB, ensure the device is already connected to the machine where you intend to flash it before triggering this.","label":"Reset","name":"reset","property":"reset","type":"enum","values":["reset"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"Silabs series 2 router","options":[],"supports_ota":false,"vendor":"Silabs"},"friendly_name":"0x84b4dbfffefceff7","ieee_address":"0x84b4dbfffefceff7","status":"successful","supported":true},"type":"device_interview"}'
[2024-12-04 19:19:29] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/button/0x84b4dbfffefceff7/reset/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x84b4dbfffefceff7/set/reset","device":{"identifiers":["zigbee2mqtt_0x84b4dbfffefceff7"],"manufacturer":"Silabs","model":"Silabs series 2 adapter with router firmware (Silabs series 2 router)","name":"0x84b4dbfffefceff7","sw_version":"2024.6.2_Nerivec","via_device":"zigbee2mqtt_bridge_0x8c65a3fffe54ba85"},"name":"Reset","object_id":"0x84b4dbfffefceff7_reset","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"payload_press":"reset","unique_id":"0x84b4dbfffefceff7_reset_zigbee2mqtt"}'
[2024-12-04 19:19:29] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x84b4dbfffefceff7/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84b4dbfffefceff7"],"manufacturer":"Silabs","model":"Silabs series 2 adapter with router firmware (Silabs series 2 router)","name":"0x84b4dbfffefceff7","sw_version":"2024.6.2_Nerivec","via_device":"zigbee2mqtt_bridge_0x8c65a3fffe54ba85"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x84b4dbfffefceff7_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x84b4dbfffefceff7","unique_id":"0x84b4dbfffefceff7_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[2024-12-04 19:19:29] error: 	z2m: Failed to configure '0x84b4dbfffefceff7', attempt 3 (Error: Bind 0x84b4dbfffefceff7/1 genBasic from '0x8c65a3fffe54ba85/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
```

antst, 2024-12-04:
> And you did the NVM3 clearing before re-flashing? It seems like your router has some left overs from past networks, simply re-flashing will not clear out the tables.

I beat you to this conclusion :) Already did full App cleaning and refreshing. So far seems to be OK :)

MilesTEG1, 2024-12-04:
> Already did full App cleaning and refreshing

would you be kind and explain how you did this cleaning?
It may be useful some day 😊

MilesTEG1, 2024-12-04:
> Already did full App cleaning and refreshing

Would you be kind and explain how you did this cleaning?
It may be useful some day 😊

antst, 2024-12-04:
> > Already did full App cleaning and refreshing
> 
> Would you be kind and explain how you did this cleaning? It may be useful some day 😊

https://github.com/Nerivec/silabs-firmware-recovery
Here you grab firmware for your device for either NVM3 cleaning or fill App cleaning. It is all explained im readme.

Nerivec, 2024-12-14:
Same as https://github.com/Nerivec/silabs-firmware-builder/issues/19
Added a note in https://github.com/Nerivec/silabs-firmware-builder/discussions/16 about it.

closed 2024-12-14 by Nerivec

Nerivec, 2024-12-15:
Saw you comment in the other thread, is this still happening after the NVM3 clear?

antst, 2024-12-15:
ah, damn, I forgot, it is gone! ))))

Nerivec, 2024-12-15:
Probably should find a way to clear the zigbee tokens when changing firmware type to avoid this type of issue in the future.

antst, 2024-12-15:
I'd suspect this is problem everyone, who was running 6.x router, will encounter.
But I also suspect, that unless you have surplus of time, you just document it :)
It is not big deal to reset. And, face reality, for most of it, this issue will be encounter by "power users", who can clear manually.