Nerivec/silabs-firmware-builder #20: Error when trying to reconfigure router in z2m

mamrai1, 2024-12-13:
This error comes up when trying to reconfigure sonoff zigbee e flashed with latest firmware  :
Error 2024-12-13 23:47:33z2m: Failed to configure 'Router', attempt 2 (Error: ZCL command 0xe0798dfffebb47c0/1 genBasic.configReport([{"minimumReportInterval":3600,"maximumReportInterval":14400,"reportableChange":0,"attribute":"modelId"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INSUFFICIENT_SPACE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.0.3/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.0.3/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.0.3/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.4.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:17) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.4.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:207:17) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.4.0/node_modules/zigbee-herdsman-converters/src/index.ts:257:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

Nerivec, 2024-12-14:
Can you try clearing the NVM3 on the router, then re-pair it in Z2M?

mamrai1, 2024-12-14:
Yes that fixed the problem

closed 2024-12-14 by mamrai1