Nerivec/silabs-firmware-builder #22: add xiao mg24

alinelena, 2024-12-30:
i can see xiao mg24 on here but no firmware for it
https://www.zigbee2mqtt.io/guide/adapters/emberznet.html

would be nice to have support for it.

ps. i have the hw and happy to help with testing...

alinelena, 2025-01-01:
thank you! that was fast I will give it a try in few days when I am back at home.

Nerivec, 2025-01-01:
Creating a manifest for it was easy, but the builds themselves are unlikely to work properly. 😅
I haven't done much research on this device, but seems it doesn't pass the serial to the USB? That would prevent the coordinator firmware from working "as-is" (might be possible to add a chip for it though, but that's extra work on user-end). The router build might be ok, but as mentioned before, needs extensive testing, that platform is very different from a ready-made device.

I'll leave the PR open until we get definitive answers to these questions, please comment on there with your findings, and any suggestion you might have.

closed 2025-01-01 by Nerivec

alinelena, 2025-01-03:
just a quick update.

managed to flash the bootloader... (quick and dirty from arduino install)
```bash
"/home/drFaustroll/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd" -d2 -s "/home/drFaustroll/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/share/openocd/scripts/" -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg -c "init; reset_config srst_nogate; reset halt; program {./seeedstudio-xiaomg24-bootloader_3.0.0.hex}; reset; exit"
```

managed to flash the router

```bash
universal-silabs-flasher --device /dev/ttyACM0 flash --firmware seeedstudio-xiaomg24-zigbee_router_8.1.0.0_115200_sw_flow.gbl                            

```

and with that i managed to have it recognized by zha and i see a signature

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0008",
      "input_clusters": [
        "0x0000",
        "0x0003"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "SeeedStudio",
  "model": "XiaoMG24",
  "class": "zigpy.device.Device"
}
```

now no idea if that is correct.

for the coordinator and answer to your first question I have no idea

when i plug it in... i get

```sh
[11319.314320] [  T28036] pcieport 0000:04:02.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 39] add_size 200000 add_align 100000
[11319.314327] [  T28036] pcieport 0000:04:02.0: bridge window [mem size 0x00200000 64bit pref]: can't assign; no space
[11319.314329] [  T28036] pcieport 0000:04:02.0: bridge window [mem size 0x00200000 64bit pref]: failed to assign
[11319.314331] [  T28036] pcieport 0000:04:02.0: bridge window [mem size 0x00200000 64bit pref]: can't assign; no space
[11319.314333] [  T28036] pcieport 0000:04:02.0: bridge window [mem size 0x00200000 64bit pref]: failed to assign
[11319.540460] [  T28485] usb 3-2: new full-speed USB device number 2 using xhci_hcd
[11319.672029] [  T28485] usb 3-2: New USB device found, idVendor=2886, idProduct=0062, bcdDevice= 1.00
[11319.672046] [  T28485] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11319.672053] [  T28485] usb 3-2: Product: Seeed Studio XIAO MG24 (Sense) CMSIS-DAP
[11319.672059] [  T28485] usb 3-2: Manufacturer: Seeed Studio
[11319.672064] [  T28485] usb 3-2: SerialNumber: CDB3E569
[11319.677317] [  T28485] input: Seeed Studio Seeed Studio XIAO MG24 (Sense) CMSIS-DAP as /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-2/3-2:1.0/0003:2886:0062.0010/input/input43
[11319.677732] [  T28485] hid-generic 0003:2886:0062.0010: input,hidraw2: USB HID v1.11 Device [Seeed Studio Seeed Studio XIAO MG24 (Sense) CMSIS-DAP] on usb-0000:39:00.0-2/input0
[11319.679640] [  T28485] cdc_acm 3-2:1.2: ttyACM0: USB ACM device
```

bad news is if I try to flash again seems to fail to find the bootloader, may need some special reset?

```bash
 universal-silabs-flasher --device /dev/ttyACM0 flash --firmware seeedstudio-xiaomg24-zigbee_ncp_8.1.0.0_115200_sw_flow.gbl                                       1 ⨯ 03/01/25 17:23:15
2025-01-03 17:23:55.339 abaddon universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=2, sdk_version='2024.12.0', ezsp_version='8.1.0.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant='sw_flow', baudrate=115200)
2025-01-03 17:23:55.339 abaddon universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-01-03 17:23:57.347 abaddon universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
```

alinelena, 2025-01-03:
ok in the end a brute force erase and burn the bootloader again fixed the issue.

```bash
"/home/drFaustroll/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd" -d2 -s "/home/drFaustroll/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/share/openocd/scripts/" -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg -c "init; reset_config srst_nogate; reset halt; flash erase_sector 0 1 last; exit"
```

of course does not work as coordinator getting some errors like this

```
[2025-01-03 19:15:49] info:     z2m: Logging to console, file (filename: log.log)
[2025-01-03 19:15:49] info:     z2m: Starting Zigbee2MQTT version 1.42.0 (commit #861cba6)
[2025-01-03 19:15:49] info:     z2m: Starting zigbee-herdsman (2.1.9)
[2025-01-03 19:15:49] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-03 19:15:49] info:     zh:zstack:znp: Serialport opened
[2025-01-03 19:15:50] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-03 19:15:51] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-03 19:16:09] error:    z2m: Error while starting zigbee-herdsman
[2025-01-03 19:16:09] error:    z2m: Failed to start zigbee
[2025-01-03 19:16:09] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2025-01-03 19:16:09] error:    z2m: Exiting...
[2025-01-03 19:16:10] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

Nerivec, 2025-01-03:
Router looks alright. I don't know ZHA well though, but shown data is in line with what I would expect.

Coordinator firmware, with Z2M, you definitely need to specify port + adapter in config. The board is not known, so it is trying to load zstack by default, which, obviously, will not work (_sidenote: this behavior changed in 2.0.0 and will error-out if unknown_). But as mentioned, it likely won't work anyway without a serial-to-usb module added to the board (if possible?).

alinelena, 2025-01-04:
indeed my bad, forgot to add adapter: ember

as expected fails. no idea if possibte to add a serial to usb module, I am not very versed in this part of the electronic. still unclear to me what do they mean by custom in z2m documentation.

```bash
[2025-01-03 23:56:08] info:     z2m: Logging to console, file (filename: log.log)
[2025-01-03 23:56:08] info:     z2m: Starting Zigbee2MQTT version 1.42.0 (commit #861cba6)
[2025-01-03 23:56:08] info:     z2m: Starting zigbee-herdsman (2.1.9)
[2025-01-03 23:56:08] info:     zh:ember: Using default stack config.
[2025-01-03 23:56:08] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-01-03 23:56:08] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-01-03 23:56:08] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-03 23:56:08] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-03 23:56:08] info:     zh:ember:uart:ash: Serial port opened
[2025-01-03 23:56:08] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-01-03 23:56:10] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-03 23:56:10] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-01-03 23:56:13] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-03 23:56:13] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-01-03 23:56:15] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-03 23:56:15] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-01-03 23:56:18] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-03 23:56:18] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-01-03 23:56:20] error:    z2m: Error while starting zigbee-herdsman
[2025-01-03 23:56:20] error:    z2m: Failed to start zigbee
[2025-01-03 23:56:20] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2025-01-03 23:56:20] error:    z2m: Exiting...
[2025-01-03 23:56:20] error:    z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

```