Nerivec/silabs-firmware-builder #25: How do I actually pair my router-flashed USB Stick to zigbee2mqtt?

FrozenGalaxy, 2025-01-04:
I successfully flashed my easyiot v1.2 stick with:
"easyiot_zb-gw04-1v2_zigbee_router_8.0.2.0_115200_sw_flow.gbl"
and updated its BL to "easyiot_zb-gw04_bootloader_2.5.2.gbl"

I checked via serial:
```
Gecko Bootloader v2.05.02
1. upload gbl
2. run
3. ebl info
BL > Reset info: 0x03 (EXT)

SL_STATUS_NETWORK_UP 0xAA6C

NWK Steering stack status 0x15

Extended Reset info: 0x301 (PIN)
RX: ZDO, command 0x0013, status: 0x00

Device Announce: 0xAA6C
```

Seems to work, I then Hit permit join in zigbee2mqtt but nothing shows up, I am using zigbee2mqtt dev (e.g 2.0+)

Nerivec, 2025-01-04:
Try to clear the NVM3, you may have some previous network remnants that's preventing it from working properly. Then pair it like a normal device (permit join). It tries to pair every 10sec as long as it's not joined.
https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3

FrozenGalaxy, 2025-01-04:
How do I clear nvm?

https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3

Nerivec, 2025-01-04:
See the link I mentioned.
If you don't want to use Ember ZLI, you can get the [GBL file](https://github.com/Nerivec/silabs-firmware-recovery/blob/main/gbl-builds/EFR32MG21A020F768IM32_nvm3_clear_0_786432_8192_16384_32768.gbl) directly and flash it like a regular firmware.

FrozenGalaxy, 2025-01-04:
Sorry, overlooked the link while on Phone, clearing the nvm worked! Thanks!

closed 2025-01-04 by FrozenGalaxy