Nerivec/silabs-firmware-builder #30: What firmware for SMLIGHT SLZB-06MG24 ?

apydo, 2025-02-06:
Hello,

I'd like to upgrade my stock zigbee firmware on SLZB-06MG24 to 8.0.2.
Is there any compatible firmware version for this device ?

apydo, 2025-02-06:
Meantime I found a working firmware from @darkxst. It was added recently.
https://github.com/darkxst/silabs-firmware-builder/pull/180
I tested this version with few devices and Z2M 2.1.0 and it seems to work well.
I hope this help.

Do you want me to test your version anyway ? I can only test coordinator version though.

Nerivec, 2025-02-06:
Yes, it should pretty much be the same for coordinator.
Router is a bit different from darkxst though.

darkxst, 2025-02-07:
> I'd like to upgrade my stock zigbee firmware on SLZB-06MG24 to 8.0.2.

I built the factory firmware, you wont change much by upgrading it now.

Nerivec, 2025-02-08:
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update6

closed 2025-02-08 by Nerivec

apydo, 2025-02-10:
I tested smlight_slzb06Mg24_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl and it seems to work. My test was on a small z2m setup with only two power outlets. Can't see any error in z2m.

apydo, 2025-02-10:
Please note that the above firmware works with software flow control (as stated in the filename).
But @darkxst's and genuine smlight firmwares work with hardware control enabled.

Nerivec, 2025-02-10:
SLZB06MG24 is software flow control only.

@darkxst variant is missing in the manifest so it's not added to filename.
https://github.com/darkxst/silabs-firmware-builder/blob/d11562d722c59805e48d72b98bd4fec7d30c3d1b/manifests/smlight/slzb06m_zigbee_ncp.yaml#L12
https://github.com/darkxst/silabs-firmware-builder/blob/d11562d722c59805e48d72b98bd4fec7d30c3d1b/manifests/smlight/slzb06Mg24_zigbee_ncp.yaml#L12