Nerivec/silabs-firmware-builder #31: Error: Failed to probe running application type

palmitoto, 2025-02-16:
Hello, I’m reaching out for your help.

I just updated the firmware of my SkyConnect with the firmware nabucasa_skyconnect_zigbee_router_8.0.2.0_115200_sw_flow.gbl from the release v2024.6.2-update5. The process went smoothly, but since then, neither Home Assistant nor Universal Silabs Flasher can communicate with it.

```
universal-silabs-flasher --device /dev/cu.SLAB_USBtoUART probe
2025-02-16 18:33:24.499 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-16 18:33:26.620 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-02-16 18:33:31.109 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2025-02-16 18:33:35.542 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2025-02-16 18:33:39.975 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-02-16 18:33:45.154 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.SPINEL at 460800 baud
2025-02-16 18:33:49.642 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.ROUTER at 115200 baud
Error: Failed to probe running application type

universal-silabs-flasher --device /dev/cu.SLAB_USBtoUART --probe-method bootloader flash --firmware ~/Downloads/nabu.gbl

2025-02-16 15:29:19.051 MBP-de-Palmier.localdomain universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=2, sdk_version='2024.6.2', ezsp_version='8.0.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_ROUTER: 'zigbee_router'>, fw_variant='sw_flow', baudrate=115200)
2025-02-16 15:29:19.051 MBP-de-Palmier.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
Error: Failed to probe running application type
```

Thanks for your help.

Nerivec, 2025-02-16:
See https://github.com/Nerivec/ember-zli/issues/14#issuecomment-2587731261 same process applies for universal-silabs-flasher or ember-zli.

closed 2025-02-17 by Nerivec

palmitoto, 2025-02-17:
Thank you @Nerivec