Nerivec/silabs-firmware-builder #43: Simplicity SDK / Zigbee EmberZNet SDK versions

jorisvervuurt, 2025-04-05:
Hey @Nerivec ,

I noticed you have a sisdk-2024.12.2 branch:
https://github.com/Nerivec/silabs-firmware-builder/tree/sisdk-2024.12.2

Simplicity SDK 2024.12.2 includes Zigbee EmberZNet SDK version 8.1.2.0:
https://github.com/SiliconLabs/simplicity_sdk/releases

In the most recent build action for your sisdk-2024.12.2 branch, I noticed that the firmwares all have `8.1.1.0` in the filename.
I guess that's a typo? `8.1.1.0` corresponds to Simplicity SDK 2024.12.1, based on the above releases page.

Or is there another reason? :)

Thanks again!

Best regards,
Joris Vervuurt

Nerivec, 2025-04-05:
Yes, looks like Silabs forgot to update the internal metadata here (and pretty much everywhere else for ZigBee...): https://github.com/SiliconLabs/simplicity_sdk/blob/1f19b3719cc750efd6ffa6604dc1c13a94eb9f11/protocol/zigbee/esf.properties#L12 that's what is used to set the firmware metadata automatically by the builder.

_I haven't had time to test any of it, last version was too buggy, never created a release, looks like this new one is off to a bad start too..._

jorisvervuurt, 2025-04-05:
Interesting; thanks for the explanation.

I've been running the build from December 29th ( https://github.com/Nerivec/silabs-firmware-builder/actions/runs/12538101751 I believe) and it's been rock-solid. I'll wait a bit then, since you said that the previous versions was buggy.

Nerivec, 2025-04-05:
Biggest problem was software flow control, that was entirely broken. Hardware flow control ones worked.

You can see the issues I reported to Silabs if you want more details: https://github.com/SiliconLabs/simplicity_sdk/issues?q=is%3Aissue%20state%3Aopen%20author%3ANerivec

Nerivec, 2025-04-13:
Unfortunately, the sw flow control issues are still present. Apparently won't be fixed until next quarter...
So, no release will go out for this one either. 🤕

jorisvervuurt, 2025-04-15:
Thanks for the update. Sad to hear Silabs isn't interested in the problems that much...

closed 2025-05-23 by github-actions[bot]