Nerivec/silabs-firmware-builder #45: Latest Silabs EmberZNet PRO Zigbee Protocol Stack Software

mamrai1, 2025-06-21:
Any plans for new firmware based on version 8.2.0.0 or 8.0.3.0?

Nerivec, 2025-06-21:
Previous version had too many problems, I never did an official release for it.
I haven't looked into the last one yet, released a few days ago (2025.6.0), it's on my todo list.

Nerivec, 2025-06-22:
Since the gap between the last well-tested version and the latest version is widening, I've create a pre-release:
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2025.6.0-pre0
But as mentioned, the same problems as described in https://github.com/Nerivec/silabs-firmware-builder/issues/43 are still present, so, not all adapters are available...

CC: @jorisvervuurt

jorisvervuurt, 2025-06-22:
@Nerivec Cool, I'll see if I can get to testing this next week. By the way, I did notice that Silabs closed some issues regarding the metadata so I'd assume that those issues have been fixed.

mamrai1, 2025-06-22:
How about the problems with the software flow control? Is it solved? Should I try it on the sonoff stick?

jorisvervuurt, 2025-07-02:
> Since the gap between the last well-tested version and the latest version is widening, I've create a pre-release (allows some testing in Z2M in case anything there needs fixing):
> https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2025.6.0-pre0

Is this still the latest release based on the latest Silabs SDK? I have just upgraded Z2M to the latest stable release which should include support. I can test coordinator firmware (with HW flow control) and router firmwares for SLZB-07 MG24 next weekend. :)

Nerivec, 2025-07-02:
Yes. I haven't had much time to test it myself... need some stability feedback for sure.

Nerivec, 2025-07-06:
Looks good. I don't think you'd see much trouble with the router firmware in any case (ironically, the underlaying app is likely far better tested than coordinator by silabs since it would affect a vast amount of devices out there - it's the same building blocks of regular routers/end devices)

jorisvervuurt, 2025-07-06:
Yep, I figured that would be the case. I'll flash all other router dongles as well as the coordinator later today. Fingers crossed. 😄

jorisvervuurt, 2025-07-11:
Still no issues to report; network seems to be very stable with these coordinator and router firmwares!

I have created a new issue for my feature request: #47